maxspahn/gym_envs_urdf #2: Error dt not accessible in mobile_reacher example

alxschwrz, 2021-11-09:
When trying to run the example: `mobile_reacher.py`, I receive the following error after the pybullet environment is built:
 "attempted to get missing private attribute '{}'".format(name)
AttributeError: attempted to get missing private attribute '_dt'

Similar issue as in the [gym_envs_planar issue](https://github.com/maxspahn/gym_envs_planar/pull/2)

This issue does affect all examples and environments.

alxschwrz, 2021-11-10:
I'll work on this.

cpezzato, 2021-11-15:
This is also an issue when trying other examples like `nLinkUrdf_reacher.py`, and `albert.py`. 

It seems the error comes from [here](https://github.com/maxspahn/gym_envs_urdf/blob/f39dcbf32678c6320e573f4dcf373a7f056c7dd5/examples/nLinkUrdf_reacher.py#L19) for the `nLinkUrdf_reacher.py`.

closed 2021-11-16 by maxspahn