maxspahn/gym_envs_urdf #3: Add basic lidar sensor

maxspahn, 2021-11-13:
It would be good to integrate a module for a lidar sensor that could be used by all robots.

Structure:
1) Folder with sensors next to the robots. In the long run all sensors should be defined there.
2) Then, every robot could add the lidar sensor, effecting the observation space.
3) In the individual robot implementations (in resources), the get_observation method would loop through all sensors to add their observations.

For lidar implementation, I suggest to use pybullets rayTest function, see [Pybullet Python API](https://usermanual.wiki/Document/pybullet20quickstart20guide.479068914/view)

alxschwrz, 2021-11-15:
**Import Error** when using the function `initSim()`:

The function `initSim()` in [generic_env](https://github.com/maxspahn/gym_envs_urdf/blob/master/pointRobotUrdf/envs/genericEnv.py) of the pointRobotURDF environment is missing the import "pybullet" in line 61. 
Changing `self._p = bullet_client.BulletClient(connection_mode=pybullet.GUI)`
to `self._p = bullet_client.BulletClient(connection_mode=p.GUI)` should resolve this.

maxspahn, 2021-11-15:
For next time use a permalink to the line [initSim()](https://github.com/maxspahn/gym_envs_urdf/blob/f39dcbf32678c6320e573f4dcf373a7f056c7dd5/pointRobotUrdf/envs/genericEnv.py#L68)
That will make sure that it actually refers to the right spot and it is easier to find.

maxspahn, 2021-11-15:
Moved to separate issue.

closed 2021-11-15 by maxspahn