maxspahn/gym_envs_urdf #6: Import Error** when using the function `initSim()` pointRobot

maxspahn, 2021-11-15:
**Import Error** when using the function `initSim()`:

The function `initSim()` in [generic_env](https://github.com/maxspahn/gym_envs_urdf/blob/master/pointRobotUrdf/envs/genericEnv.py) of the pointRobotURDF environment is missing the import "pybullet" in line 61. 
Changing `self._p = bullet_client.BulletClient(connection_mode=pybullet.GUI)`
to `self._p = bullet_client.BulletClient(connection_mode=p.GUI)` should resolve this.

_Originally posted by @alxschwrz in https://github.com/maxspahn/gym_envs_urdf/issues/3#issuecomment-968720930_

maxspahn, 2021-11-15:
This problem is also true for most other environments, see [mobileReacher](https://github.com/maxspahn/gym_envs_urdf/blob/f39dcbf32678c6320e573f4dcf373a7f056c7dd5/mobileReacher/envs/vel.py#L65) for example.

maxspahn, 2021-11-15:
The function initSim() is actually not used in the environments. 
Should be removed entirely. Then this issue does not need solving.

alxschwrz, 2021-11-16:
Should the function initSim() be removed entirely from all the environments?

maxspahn, 2021-11-17:
Yes, that would be ideal.

maxspahn, 2021-11-24:
It might be solved by working on #17

maxspahn, 2021-11-25:
closed by #18

closed 2021-11-25 by maxspahn