maxspahn/gym_envs_urdf #29: Upload pip package

maxspahn, 2021-12-09:
Eventually, we should upload this package to the official pip repositories.
Then the installation would be even simpler and we could attract some more users.

maxspahn, 2022-02-21:
Adding a license should be done before, see [ft-add-license](/maxspahn/gym_envs_urdf/tree/ft-add-license)

maxspahn, 2022-04-12:
Let's schedule the first release for 25 April!
What do you think @GijsGroote @alxschwrz ?

alxschwrz, 2022-04-12:
Please excuse my late reply. I somehow did not see I was mentioned in this issue.

**Scope:** Is the goal to upload both packages [gym_envs_planar](https://github.com/maxspahn/gym_envs_planar) as well as [gym_envs_urdf](https://github.com/maxspahn/gym_envs_urdf) as official pip repos at the same time? Speaking about gym_envs_planar, I think the 25 April should be feasible.

**Regarding RL**: Is the target group also mostly RL users? If this is the case, we might need to make some major changes to the repos before the release (e.g. provide a reasonable reward function, method of sampling goals inside the reset-function, terminal conditions, normalization of observation and action spaces, maybe providing some baselines). If this is the case, the 25th might be quite soon. However this could also be subject to a future release version, right? What's your take on that? I believe the repositories might also be a great contribution to the _RL in Motion Planning_ community as a quite challenging set of tasks.

maxspahn, 2022-04-12:
> Scope:

Ideally, upload both at the same time, but that seems very challenging.

> Regarding RL:

No, RL should be one potential application, but not the main target group. Also, it seems very difficult to formulate a generic reward function. Maybe, we could propose some rewards functions in a future release, but I don't see the direct need now.

maxspahn, 2022-04-12:
I added a project for this. Then we can list all these comments in there as todos

closed 2023-01-04 by maxspahn