maxspahn/gym_envs_urdf #57: Automated extraction of joint ids

maxspahn, 2022-03-02:
Joint indices for reading in the limits, controlling the joints, and disable castor wheels is hard-coded at the moment. 
It would be much better if this was done automatically, or at least by name.

A starting point could be:
https://github.com/maxspahn/gym_envs_urdf/blob/d4d49dc1bb7486f5ac006efaf79e8ce08603d7c9/urdfenvs/tiagoReacher/resources/tiagoRobot.py#L23-L53

maxspahn, 2022-04-22:
A possible solution to this can be found in 969cbd8ef3488027d55803bf8cf10aea1eb0b0e0

SkyloveQiu, 2022-04-27:
That possible solution could not be used, since the connection with the server is not established when the code called `set_joint_indices` function.

maxspahn, 2022-04-28:
Could we change that?
It is true, that the connection `p.loadURDF` must be called before we can access the joint names. Maybe, the joint indices could be extracted right after that function call in the reset function.

SkyloveQiu, 2022-04-28:
Can I move the `loadURDF` to the `__init__` function?

SkyloveQiu, 2022-04-28:
it will give error in `get_observation_space` function that object has no attribute `get_observation_space`

SkyloveQiu, 2022-04-28:
I think we can add `set_joint_indices` before we call `get_observation_space` and it is working.

PR #91 (2022-05-10): Ft automated extraction joint
The first version is finished, resolve #57.
There are a few problems, to make the extraction of id, I introduce preload idea which means loading the robot before generating the environment. 
I have noticed that the preload will slow down the start-up process.
closed 2022-05-10 by maxspahn