maxspahn/gym_envs_urdf #59: Running on Windows

RJMHK, 2022-03-09:
Here are the steps to run the code on windows (with spyder):
- Download Git for Windows: https://git-scm.com/download/win
- Open Git Bash
- Direct to a folder where you want the code to be stored
- Copy 
- Run the command _git clone_ https://github.com/maxspahn/gym_envs_urdf.git
Now the files are in the folder and linked to the github. To run the files:
- Open spyder
- In the console, direct to the folder you made earlier, and one step further to _gym_envs_urdf\_
- run _pip install ._
Now you should be able to run the codes found in the folders

maxspahn, 2022-03-09:
Thanks @RJMHK 
I will add this to the documentation soon.

maxspahn, 2022-04-03:
Have you tried it with the poetry installation?

maxspahn, 2022-04-03:
Can you maybe comment on 2a8b78d, @rensheemskerk

maxspahn, 2022-04-22:
Documentation added in #81

maxspahn, 2023-06-09:
Closed with docs added a while ago.

closed 2023-06-09 by maxspahn