maxspahn/gym_envs_urdf #74: Supported Python versions

maxspahn, 2022-04-03:
The pyproject.toml file suggests that python 3.6 until 3.10 are supported.
However, when installing it with python3.10, I run into trouble because of numpy package.
Can this be confirmed by anyone? Or maybe somebody has it running with python 3.10.

maxspahn, 2022-04-03:
There also seems to be a problem with casadi 3.5.5: https://github.com/casadi/casadi/issues/2877

GijsGroote, 2022-04-13:
I would prefer python 3.10 because it has a match/case pattern (I like).
Would it be okey to force user to use python 3.10? 
Or do we want to allow python 3.6 - 3.9 ?

maxspahn, 2022-04-13:
We have to allow 3.6.2 at least, because ros melodic uses python 3.6.2 at max.
We can change this as soon as we update all the platforms to newer ros versions.

SkyloveQiu, 2022-04-25:
I have experienced the same problems for python3.10, I have observed that numpy-1.19.5 is not supported by the python3.10.
An update for numpy is needed.

maxspahn, 2022-04-25:
If you remove the file `poetry.lock`, you need to run `poetry update` to create a new lock for the installation.
Can you check what is the oldest numpy version supported in python3.10? Maybe this version is then already supported in python3.6.

This would allow us to keep both, python3.6 and python3.10.

If this proves to be insufficient, we can fall back to using different dependencies depending on the python version.
Let me know if you need help with that.

SkyloveQiu, 2022-04-25:
The oldest version NumPy version supported in python 3.10 is 1.21.x, which only supports python3.7+. Keeping both is not possible at this moment. It's better to use different dependencies on the python versions.
And I have observed that Mac Os does not support pybullet. I am not sure why but the installation failed.

SkyloveQiu, 2022-04-25:
Problems has been mentioned here : numpy/numpy#17044

SkyloveQiu, 2022-04-25:
With Ubuntu 22.04, Users have to downgrade their python version to 3.9, this [tutorial](https://www.linuxcapable.com/how-to-install-python-3-9-on-ubuntu-22-04-lts/) should help.

maxspahn, 2022-04-26:
We should try to work with conditional versions like suggested in https://github.com/maxspahn/motion_planning_scenes/blob/main/pyproject.toml.
There is a newer casadi version (3.5.6.pre2) that supports python 3.10.

SkyloveQiu, 2022-04-26:
I have tried that, however there is one SolverProblemError.
`Because motion-planning-scenes (0.1.1) depends on casadi (<=3.5.5)
   and no versions of motion-planning-scenes match >0.1.1,<0.2.0, motion-planning-scenes (>=0.1.1,<0.2.0) requires casadi (<=3.5.5).
  So, because urdfenvs depends on both casadi (3.5.6.pre2) and motion-planning-scenes (^0.1.1), version solving failed.`
Could you update the Motion-planning-scenses on pypi?

maxspahn, 2022-04-26:
It is wise to not spam the pypi index in such an early phase of the software. Let's change the dependency to a direct github dependency instead for now. Could you create a PR for that?

SkyloveQiu, 2022-04-29:
Problems are not resolved, still can not install on python3.10

maxspahn, 2022-04-29:
What is the error? Can you show the whole output of poetry install?

SkyloveQiu, 2022-04-29:
it automatically installs numpy 1.19.5 which does not support python3.10

SkyloveQiu, 2022-04-29:
we can revert https://github.com/maxspahn/motion_planning_scenes/pull/19/commits/6b69e66a3418e22ed787f771ec7e5331251be8bb and use the old setting. That works for python3.10, or we can just remove support for python3.10 until further notice.

maxspahn, 2022-04-29:
That broke the installation of other python versions. So let's rather remove support for python3.10 until I find a solution.

SkyloveQiu, 2022-04-29:
Ok, then we can refer this  [tutorial](https://www.linuxcapable.com/how-to-install-python-3-9-on-ubuntu-22-04-lts/) for people who uses ubuntu 22.04

maxspahn, 2022-04-29:
sure!

halvorak, 2022-08-31:
Any updates when Casadi is available fo python 3.10? On https://web.casadi.org/get/ it still says casadi works until 3.9.

SkyloveQiu, 2022-08-31:
@halvorak Dear Halvorak, 
    we are not sure when will the Casadi be avaliable for python 3.10, if you need to run this project, I would suggest you to  follow this  [tutorial](https://www.linuxcapable.com/how-to-install-python-3-9-on-ubuntu-22-04-lts/) to downgrade your python version to 3.9. 

Have a nice day!
SkyloveQiu

halvorak, 2022-12-07:
Hi @SkyloveQiu, normally that would not be a problem but I need to use python 3.10 due to a different package. Based on this (https://github.com/casadi/casadi/issues/2877), it seems like the support for python 3.10 is available in the nightly build? I checked https://github.com/casadi/casadi/releases/tag/nightly-develop, but I'm not really sure which of those to run? I'm running on windows.

SkyloveQiu, 2022-12-08:
Hi, @halvorak you usually can change the version under [pyproject.toml] https://github.com/maxspahn/gym_envs_urdf/blob/master/pyproject.toml)
you can add the specific version for yourself; in the short term, we will step toward python3.10, but the timeline is still unknown. If you have any problems, please let me know.

maxspahn, 2023-01-05:
In #143 , we moved to support python3.10 at the cost of no longer supporting pyhon 3.6 and python3.7. It is going to be pushed to master (and thus to pypi) soon.