maxspahn/gym_envs_urdf #77: Placing walls in the environment should be more intuitive

GijsGroote, 2022-04-11:
When walls should be places  other than the default, it takes to long to understand how to place walls.

maxspahn, 2022-04-12:
What would you find most intuitive?

GijsGroote, 2022-04-12:
TL;DR
remove `add_walls`, extend `add_obstacle(obstacle)` to `add_obstacles(obstacle, positions=list[pose_2D, ....])`

Let's take into account the implementation of obstacles, there are 2 methods to create walls:
``createCollisionShape`` or ``createMultiBody`` 
or
``loadURDF / loadSDF / loadMJCF``

the [quickstart documentation](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.jgm6tud6blu3) already suggest it is better to load in urdf files with the `loadURDF` method. So I suggest placing walls should be upgraded from createMultiBody to loadURDF. 

This would mean the `add_walls` would become exactly as `add_obstacle`
additionally it would be useful to place walls on multiple pose_2D locations.  

where add_walls(wall_urdf, locations = list[pose_2D, ....])
where numpy array pose_2D = (x, y, orientation) of the center of mass of the wall. 
(what do you think, pose_2D should point to center of mass, corner of wall, something else?)

* easy to place walls with pose_2D
* This gives the opportunity to easily place the same wall on multiple locations
* This makes every object in the simulation an obstacle (no more collisionshape's)

It would be logical to recall `add_walls` to `add_multiple_obstacles` or to fully intergrate `add_walls` into `add_obstacle`.

maxspahn, 2022-04-12:
alright, I'll have a look at it!

closed 2022-05-10 by maxspahn