maxspahn/gym_envs_urdf #100: Missing package 'wheel' for depedency urdfpy

c-salmi, 2022-07-21:
Tested the pip install on a clean virtual environment, but the `wheel` package dependency is missing in [urdfpy/fix-networkx-dependency](https://github.com/maxspahn/urdfpy/tree/fix-networkx-dependency). 

Some info to reproduce the error:

python version: 3.8.10
pip version: 20.0.2
OS: ubuntu 20.04.4 LTS

steps:
``` bash
python3 -m venv venv # create clean virtual env
source venv/bin/activate #active virtual env
git clone git@github.com:maxspahn/gym_envs_urdf.git
cd gym_envs_urdf
pip3 install .
```

output:

[image]

maxspahn, 2022-07-21:
I can confirm this for python 3.8.9 on ubuntu 22.04

c-salmi, 2022-07-25:
This issue might have been on the wrong repository. The dependency was solved upstream in [urdfpy](https://github.com/maxspahn/urdfpy/commit/63b0b4d1f06136cbfd5367f556eb4d29f0d140b6), by you @maxspahn xD. No need to update anything in here, because it was added on the same branch `fix-networkx-dependency`.

closed 2022-07-25 by c-salmi

maxspahn, 2022-07-25:
Great to hear that my bugfixing is actually useful sometimes:D