maxspahn/gym_envs_urdf #113: createCollisionShape/VisualShape using wrong unique id

casparvv, 2022-07-28:
Method function [``add_shapes``](https://github.com/maxspahn/gym_envs_urdf/blob/94e6007353d6e17207d1d694cbfc19acc3b59ad0/urdfenvs/urdfCommon/urdf_env.py#L401-L408) of the class ``UrdfEnv`` uses the same unique id for the ``baseCollisionShapeIndex`` and ``baseVisualShapeIndex`` which can lead to problems if an obstacle or goal is defined beforehand.

An obstacle or goal only creates a [visualShape id](https://github.com/maxspahn/motion_planning_scenes/blob/355b6358d977749d07a8f12923cd2ef1ed66a314/MotionPlanningGoal/staticSubGoal.py#L130-L131) or only a [collisionShape id](https://github.com/maxspahn/motion_planning_scenes/blob/355b6358d977749d07a8f12923cd2ef1ed66a314/MotionPlanningEnv/sphereObstacle.py#L145-L146). The ``add_shapes`` function used to create a wall, for example, will create a new collisionShape id and use that id for both the collisionShape and visualShape, while the visualShape under that id is still the previous created visualShape (if it exists) showing the visual of the previous created goal.

Possible fixes: create both a visualShape and collisionShape for the goal and obstacle, both for the shapes inside ``add_shapes`` or both for all.

casparvv, 2022-07-28:
Or set [``baseVisualShapeIndex``](https://github.com/maxspahn/gym_envs_urdf/blob/94e6007353d6e17207d1d694cbfc19acc3b59ad0/urdfenvs/urdfCommon/urdf_env.py#L405) to ``-1`` in ``add_shapes``.

casparvv, 2022-08-01:
Created PR #114 to solve the issue.

maxspahn, 2022-08-04:
#114 has not been merged yet. Leave it open until it is merged.

casparvv, 2022-08-04:
Merged now!

closed 2022-08-04 by casparvv