maxspahn/gym_envs_urdf #120: Multirobot environments

maxspahn, 2022-09-27:
Some users might consider using this environment, if there were multi-robot system.

I have started a branch on this and it seems feasible. The first idea is to make use of the generic urdf environment, but using
list instead of a single urdf-file. 
Let's discuss this further in this thread @c-salmi.

The branch name is ft-multi-robots.

c-salmi, 2022-10-04:
I've added on your work with the list of genericrobots in 431cd7aafdea00ac342b7fad6ac013d06068b0c6

There, I've moved the call `apply_action` to the robot class itself instead of within the environment. This way we can simply loop over the robots and call their `apply_action`, and we support having multiple robots with different control modes at the same time, making the environment control_mode agnostic.

What do you think of this approach?

c-salmi, 2022-10-05:
The `multi_robot.py` script now works with any robot (holonomic, differentialdrive, bicyclemodel). The control mode is also specified within the robot class itself. Therefore we now need only one env called `urdf-env-v0` and pass in a list of robots (that each have their apply_action function etc.

looks something like this:
```python3
    robots = [
        GenericUrdfReacher(urdf="pointRobot.urdf", mode="vel"),
        TiagoRobot(mode="vel"),
        Prius(mode="vel")
    ]

    env = gym.make(
        "urdf-env-v0",
        dt=0.01, robots=robots, render=True
    )

```

(Note: TiagoRobot and Prius are not obsolete because, genericUrdfReacher only works with holonomic robot urdfs for now)

I propose to refactor the folder structure, since we don't need all the custom envs anymore. What do you think of a structure like this, @maxspahn ?:

    ├── ...
    ├── urdfenvs
    │   ├── robots
    │   │   ├── albert
    │   │   │   ├── albert.urdf
    │   │   │   ├── albert_robot.py 
    │   │   │   └── meshes
    │   │   └── boxer, etc.
    │   ├── urdf_common
    │   │   ├── urdf_env.py
    │   │   ├── generic_robot.py
    │   │   ├── holonomic_robot.py
    │   │   └── differential_drive_robot.py,  etc.
    └── ...

maxspahn, 2022-10-05:
Thanks @c-salmi.

I agree with the new structure. It seems much neater, although we have to make sure that the corresponding changes indicate a new version according to https://semver.org/. 

So to recap, all robots that are not holonomic (full list: albert, tiago, boxer, iris-drone, prius) must be kept so far. And for the others, we only make the mesh available through the repository?

The removal of all the different environments is very nice and we should push for that quickly.

maxspahn, 2022-10-05:
I will look at the changes an the adaptations needed in the tests soon, so that we can create a PR.

c-salmi, 2022-10-05:
Sounds good, I'll refactor it to the above structure then

maxspahn, 2022-10-12:
Solved in #122 
Integrated on master in #123

closed 2022-10-12 by maxspahn