maxspahn/gym_envs_urdf #128: Sensor observations spaces not all correctly implemented

c-salmi, 2022-10-13:
Currently, for at least the `FullSensor`, the `get_observation_space` method is not correct. 

Therefore we are currently bypassing the oberservation space check in the `_get_obs()` function of `UrdfEnv` [see](https://github.com/maxspahn/gym_envs_urdf/blob/340aea2214dfda2129708114f72c6493bd109acc/urdfenvs/urdf_common/urdf_env.py#L253).

maxspahn, 2022-12-06:
@Capsar The FullSensor should be fine. Otherwise, you might pass the trajectories by the moving obstacles directly to your local planner and skip perception entirely.

Capsar, 2022-12-06:
@maxspahn Yep, figured out how to implement the FullSensor, was stuck on what the masks meant. After taking a look at the sensor test I figured it out: https://github.com/maxspahn/gym_envs_urdf/blob/2bb2159f9f44cb3171d8b638129cf2e149f90106/tests/test_obstacle_sensor.py#L23

closed 2024-04-23 by maxspahn