maxspahn/gym_envs_urdf #134: Add Isaac Gym Environment

SkyloveQiu, 2022-11-01:


c-salmi, 2022-11-14:
Hi Qiu,

Heard you were working on this but was not able to test it due to your laptop specs not having enough gpu vram. I'm actually also interested in adding this option. Could you push your work to a dedicated branch for this new feature?

SkyloveQiu, 2022-11-14:
Hi chadi:
The gym.make function is not working, I did not find a replacement function from the documentation, do you know which function replaces the original gym.make?

SkyloveQiu, 2022-11-14:
`robots = [
        AlbertRobot(mode="vel"),
    ]
    env = gym.make(
        "urdf-env-v0",
        dt=0.01, robots=robots, render=render
    )`

The code I am referring to.

SkyloveQiu, 2022-11-14:
@c-salmi [Link](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs) this could help I believe

SkyloveQiu, 2022-11-16:
@c-salmi [start point](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/main/isaacgymenvs/tasks/base/vec_task.py) I have found this `VecTask` class which could be our start point.

c-salmi, 2022-11-16:
@SkyloveQiu Yes exactly their `step` method implementation would be a good reference to see how we can convert our [step](https://github.com/maxspahn/gym_envs_urdf/blob/2bb2159f9f44cb3171d8b638129cf2e149f90106/urdfenvs/urdf_common/urdf_env.py#L227) from pybullet to isaacgym.

SkyloveQiu, 2022-11-18:
Probably we are not going to make our step into issacgymenv, let's implement everything from scratch and build it with openai gym, more people can use our project to replace their pybullet.

maxspahn, 2023-01-06:
In a first step, let's separate the physics engine from the rest. I started doing this in https://github.com/maxspahn/gym_envs_urdf/tree/ft-separate-physics-engine

maxspahn, 2023-06-09:
Closing as this integration is put on hold in favor of integrating MuJoCo instead.

closed 2023-06-09 by maxspahn