maxspahn/gym_envs_urdf #136: Missing MotionPlanningGoal or MotionPlanningEnv

maxspahn, 2022-11-28:
Installing the bare minimum of the package leads to missing some package.
This might throw an error like:

```
warnings.warn(str(err))
Traceback (most recent call last):
  File "/home/josyula/Programs/MAS_Project/gym_envs_urdf/examples/multi_robot.py", line 51, in <module>
    run_multi_robot(render=True, obstacles=True, goal=True)
  File "/home/josyula/Programs/MAS_Project/gym_envs_urdf/examples/multi_robot.py", line 34, in run_multi_robot
    from examples.scene_objects.goal import dynamicGoal
  File "/home/josyula/Programs/MAS_Project/gym_envs_urdf/examples/scene_objects/goal.py", line 1, in <module>
    from MotionPlanningGoal.staticSubGoal import StaticSubGoal
ModuleNotFoundError: No module named 'MotionPlanningGoal'
```

Capsar, 2022-12-08:
Tried the top command, but there is a small typo: should be `pip install "urdfenvs[scenes]"`

maxspahn, 2023-01-10:
In #148, the dependency changed from optional to required, thus making the installation easier in the future.

closed 2023-01-10 by maxspahn