maxspahn/gym_envs_urdf #137: Installing pybullet on windows

maxspahn, 2022-11-28:
Many people try to use the simulation environment on Windows. 
For most people this results in errors for the package ```pybullet``` as it requires a specific c++ compiler.
Let us gather information and problems related to this in this issue.

DasWilheim, 2022-12-09:
Following this instruction on stackflow worked for me.

https://stackoverflow.com/questions/64261546/how-to-solve-error-microsoft-visual-c-14-0-or-greater-is-required-when-inst

behradkhadem, 2023-03-27:
I use WSL2 to tackle this issue. With Ubuntu subsystem you won't need a C++ compiler or else to run pybullet. You can simply just install it as a python package and use it. By using subsystems, you have flexibility of Linux with UX of windows. Also, you can run Linux GUI on Windows too, for that you will need [XLaunch](https://x.cygwin.com/docs/xlaunch/).
If you want, I can write some docs with pictures and stuff. 

[image]

maxspahn, 2023-03-27:
@behradkhadem Thanks for pointing that out! If you could write some docs, and create a PR that would be very much welcome. I don't use windows personally, so it is hard for me to test it on windows.
Let me know if you need any help, also concerning the workflow.

behradkhadem, 2023-03-27:
> @behradkhadem Thanks for pointing that out! If you could write some docs, and create a PR that would be very much welcome. I don't use windows personally, so it is hard for me to test it on windows.
> Let me know if you need any help, also concerning the workflow.

Sure, I'll look into it at the weekend!

maxspahn, 2023-06-09:
Closed as docs were updated by @behradkhadem

closed 2023-06-09 by maxspahn