maxspahn/gym_envs_urdf #138: Environment compilation failed in Linux VM with Ubuntu 20.04.4

Yope01, 2022-12-06:
After following the tutorial in the README, the installation was successful but then trying to run `python3 point_robot.py` in the examples folder I get the following error. The environment opens for a very brief moment before closing again.

`int_robot.py
pybullet build time: May 20 2022 19:44:17
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Failed to create GL 3.3 context ... using old-style GLX context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=VMware, Inc.
GL_RENDERER=SVGA3D; build: RELEASE;  LLVM;
GL_VERSION=2.1 Mesa 21.2.6
GL_SHADING_LANGUAGE_VERSION=1.20
pthread_getconcurrency()=0
Version = 2.1 Mesa 21.2.6
Vendor = VMware, Inc.
Renderer = SVGA3D; build: RELEASE;  LLVM;
Compile failed:
0:1(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, and 1.00 ES`

maxspahn, 2022-12-06:
It looks like it is related to your hardware. Maybe, you can follow this: https://discourse.vtk.org/t/error-glsl-1-50-is-not-supported/1444/5.

cagvanderaa, 2022-12-06:
Hi, we're having the same problem and the provided link didn't get it working for us. Does anybody have any suggestions for this problem?