maxspahn/gym_envs_urdf #139: Changing camera position while rendering

Capsar, 2022-12-06:
Hi,

I was wondering how one could change the initial camera setup of of the POV while rendering.
Our goal is a top view with specific coordinates in the corners of the view. I suppose this last part will be trail and error, but how does one change the initial camera POV?

maxspahn, 2022-12-06:
Currently, this API does not support this. However, you can use the pybullet function for that in your main function using the pybullet api as suggested in https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=12166#:~:text=Try%20holding%20the%20CONTROL%20or,camera%2C%20while%20dragging%20the%20mouse.

Let me know if this help and feel free to write an API in this project ;)

maxspahn, 2023-02-15:
Function `reconfigure_camera` added in #161

closed 2023-02-15 by maxspahn