maxspahn/gym_envs_urdf #179: Defining reward function for an environment

behradkhadem, 2023-04-28:
Hi everyone, hope you're all doing well.

As we know, the reward value inside `GenericRobot` class is statically set to 1. 
https://github.com/maxspahn/gym_envs_urdf/blob/be7532ae35675c5a2fd8c0d1782e8dbfd684e446/urdfenvs/urdf_common/urdf_env.py#L278
How can I define an environment with custom reward function (using data from sensors and stuff)? Upon cursory investigating the code I came across a class called `GoalComposition` from a different package, but I didn't see this class be used inside any example. Is this related to this package?

And if not, is there any page or docs for future tasks and roadmap of this package in order to know what are prerequisites to implement reward function inside this package and help it develop? Are there any plans to implement reward function at all?

maxspahn, 2023-04-28:
Hi @behradkhadem again,

good to hear from you again.
I am not a user of RL nor an expert in reward function design, so you will have to come up with something yourself.

However, it is easy to design a simple sparse reward, or even a dense reward based on the distance to the goal.

Indeed the `GoalComposition` can be used here, as you have access to the goal position and orientation potentially.

From the top of my head, I would implement in the following way.
```python
goal_position = self._goals[0].position()
cur_position = <pybullet syntax to get current position of the end-effector/robot>
reward = 1/np.linalg.norm(goal_position - cur_position)
```

It would be really nice if you could propose something here.
However, reward shaping is an entire field of research so a generic reward function for urdfenvs seems very ambitious.

Looking forward to hearing from you.

Best,@maxspahn

behradkhadem, 2023-05-07:
hello? @maxspahn

maxspahn, 2023-05-07:
Hi @behradkhadem ,

Sorry for me not being responsive on that. I am on vacation, will look into the OR next week.

Have you checked why the tests fail?

Best,
@maxspahn

behradkhadem, 2023-05-08:
Ok then, enjoy your vacation.
And tests failed because I've changed the implementation. 😅 But haven't checked it TBH and I was planning on doing that after solving the issue with sensors.

closed 2023-05-17 by behradkhadem