maxspahn/gym_envs_urdf #199: What is a goal? What are the obstacles?

behradkhadem, 2023-06-25:
Hello @maxspahn, I know that this is not an issue, but I couldn't find a better way of asking this question.

Inside the examples, there are two concepts called `Goal` (evidently dependent on a package called `mpscenes`) and `Obstacle`. What are they exactly? What are their use cases? I know that they add some data to observation dictionary but I fail to understand what are they used for? Are obstacles movable? Can a panda arm pick an obstacle? Or do I need to create new obstacles (from URDF files)? 
Docs doesn't have enough information on the subject and it just says _todo_.

maxspahn, 2023-06-26:
Hi @behradkhadem,
Obstacle are physical things, that a robot can interact with, it's either movable or not, see flag in the constructor dictionary. A goal is an abstract concept acting as a target for the motion planner, it can be visualized in the environment but does not affect the environment in a physical way.

behradkhadem, 2023-06-26:
> Hi @behradkhadem, Obstacle are physical things, that a robot can interact with, it's either movable or not, see flag in the constructor dictionary. A goal is an abstract concept acting as a target [...]

Thanks for responding. 
Does `mpscenes` have any docs or stuff?

maxspahn, 2023-06-27:
No, not at all. :(

closed 2023-06-29 by behradkhadem