maxspahn/gym_envs_urdf #223: Updates in sensor

maxspahn, 2023-08-11:
Hello Max.
I saw some improvements made on sensor but didn't have time to check the code and the commits. Could you please elaborate that what was the issue and what's being fixed right now? Because I didn't see any issues in tests that I've made and right now I'm able to train the RL agent (but failing to find a place to run the code. 😅).

Thanks.

_Originally posted by @behradkhadem in https://github.com/maxspahn/gym_envs_urdf/issues/222#issuecomment-1674493227_

maxspahn, 2023-08-11:
What is the state at which you are? Only then, I can tell you what has changed. However there is also the CHANGELOG.md on the individual releases.

Let me know what exactly your state/question is.

behradkhadem, 2023-08-11:
> What is the state at which you are? Only then, I can tell you what has changed. However there is also the CHANGELOG.md on the individual releases.
> 
> Let me know what exactly your state/question is.

I don't know what you mean by "the state" but I'm using the position of the panda robot arm to perform pick and place. And I obtain the position of the obstacle via the Full sensor.

maxspahn, 2023-08-11:
That hasn't changed recently. So it should be safe for you :wink:

behradkhadem, 2023-08-11:
> That hasn't changed recently. So it should be safe for you 😉

Pew! Thanks! 😅😅

BTW, where can I ask my questions about the package? This package doesn't have [discussions](https://github.com/features/discussions), and questions are not issues. And I see that you are meticulous about this, so I wanted to know your preference on this matter.

Thanks.

maxspahn, 2023-08-11:
> BTW, where can I ask my questions about the package? This package doesn't have [discussions](https://github.com/features/discussions), and questions are not issues. And I see that you are meticulous about this, so I wanted to know your preference on this matter.

Good point. 
I added a discussion now, see https://github.com/maxspahn/gym_envs_urdf/discussions

closed 2023-08-11 by behradkhadem