maxspahn/gym_envs_urdf #228: Add goal reaching as a terminated condition

maxspahn, 2023-08-21:
It seems a good idea, #226, to trigger a terminated state once the goal is reached if a goal is present.

behradkhadem, 2023-08-29:
I think goals are handled inside `mpscenes` and I don't have access to its code. But it's a good idea. It could have limited functionality (due to it being limited to position and velocity of things) but most RL projects in robotics require something put in some place, so I think we'll be fine!

maxspahn, 2023-08-29:
@behradkhadem, do you think you could work on that?
By the way, `mpscenes` is public, see https://github.com/maxspahn/motion_planning_scenes

behradkhadem, 2023-08-29:
I can but I don't know when I will have the free time to do that. It may be in a month or more.