maxspahn/gym_envs_urdf #230: Robot doesn't respond to action in `tor` (torque) mode

behradkhadem, 2023-09-03:
Hello @maxspahn, I think there's an issue with torque mode in robots. take this code for example: 
```
import gymnasium as gym
from urdfenvs.urdf_common.urdf_env import UrdfEnv
from urdfenvs.robots.generic_urdf import GenericUrdfReacher
from urdfenvs.sensors.full_sensor import FullSensor
from urdfenvs.scene_examples.goal import goal1
import numpy as np

robots = [
    GenericUrdfReacher(urdf="pointRobot.urdf", mode="tor"),                                 # MODE IS SET HERE!
]
render: bool = True

env = UrdfEnv(render=render, robots=robots)

env.add_goal(goal1)

sensor = FullSensor(['position'], ['position', 'size'], variance=0.0)
env.add_sensor(sensor, [0])
env.set_reward_calculator(SampleReward())
env.set_spaces()

defaultAction = np.array([ 0.16851664, -0.38915175, -0.09081227])
pos0 = np.array([0.0, 0.0, 0.0])
vel0 = np.array([0.0, 0.0, 0.0])

for _ in range(100):
    ob = env.reset(pos=pos0, vel=vel0)
    print(f"Initial observation : {ob}")
    #assert np.array_equal(initial_observations[0], ob)

    # history = []
    for _ in range(250):
        action = defaultAction
        ob, *_  = env.step(action)
```
Shouldn't robot respond to torque input? Is there something that I'm doing wrong? Everything seems fine in `vel` mode (and I even trained a simple RL agent with it) but I think there is something wrong with `tor`.

behradkhadem, 2023-09-03:
Now that I'm seeing, we have the same issue with `panda_with_gripper.urdf` too!

maxspahn, 2023-09-05:
the magnitude of the applied torques is too small. Usually, for torques to be effective and overcome the friction, they must be in the order of 10000. Then however, you might have issues with the effort limits in the urdf's.

behradkhadem, 2023-09-06:
> they must be in the order of 10000

How much?! 🤯🤯🤯
What is the unit of torque here? If it's N.m, this kind of torque would be enough for deforming a steel beam, if not break it! I don't think a robotic arm (based on my limited understanding of DC motors and stuff) shouldn't require more than 50 N.m of torque for handling stuff, and that is for big industrial ones! 
For panda arm the maximum torque that we can apply is in joint 4 which is 2.8 N.m. And if so, we shouldn't have such issues with point robot, should we? It's either an issue or we have too much friction in my opinion!

behradkhadem, 2023-09-06:
> Yeah, I agree. I never really looked into that to be honest, because the friction model is unlikely to be very accurate with respect to the real robot. It would be very nice if you could investigate that, @behradkhadem .

Sure, but a few questions.
Isn't the friction randomly generated? And is friction a property of URDF file or something generated by the physics simulator? If it's generated by our package, can you pinpoint where it's being implemented? Because I couldn't find anything. But upon my cursory look, code is fine and the torque input is being called via `setJointMotorControl2` in `holonomic_robot.py` line 89. I'll dive deep into it tomorrow. 

Thanks in advance!

maxspahn, 2023-09-06:
So, there are some parameters inside the URDF file. However, it feels like these are ignored by pybullet all together and in effect, there are some default values for all joints.

You can change the friction parameters for individual joints similar to how I did it for the castor wheels: https://github.com/maxspahn/gym_envs_urdf/blob/c7f7cb6bb22083695d625bbb20952db45574f95d/urdfenvs/urdf_common/differential_drive_robot.py#L110-L118

There is also a thread for a similar topic for a kuka arm: https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=12644

I hope this hepls a bit.