maxspahn/gym_envs_urdf #233: Forward Kinematics for Mobile Panda

behradkhadem, 2023-09-17:
Hi @maxspahn ,

This is issue is the same issue I had with panda with gripper arm (https://github.com/maxspahn/gym_envs_urdf/issues/201) but for `mobilePandaWithGripper.urdf`. I couldn't find a FK object for this robot and I presume that we should use the `PandaFK`, but again, I think that the position being returned via the code inside (https://github.com/maxspahn/gym_envs_urdf/issues/201#issuecomment-1650344088) is returning a random value. What is the correct way of getting the position of end-effector in this urdf file?

behradkhadem, 2023-09-17:
PS: I think this is an issue: 

[image]
I think the issue stems from incorrect bounds for end-effector inside URDF file, right?

maxspahn, 2023-09-22:
> I think the issue stems from incorrect bounds for end-effector inside URDF file, right?

Probably, have you tried to fix that?

> This is issue is the same issue I had with panda with gripper arm (https://github.com/maxspahn/gym_envs_urdf/issues/201) but for mobilePandaWithGripper.urdf. I think I should use MobilePandaFk for this matter, but there is no example or anything for this robot. What is the correct way of getting the position of end-effector in this urdf file? I'm looking for something like https://github.com/maxspahn/gym_envs_urdf/issues/201#issuecomment-1650344088.

The best is to use the generic_fk from the forward kinematics package, then you can directly pass your urdf file.

behradkhadem, 2023-10-08:
> Probably, have you tried to fix that?

No, first I wanted to make sure that problem stems from the URDF file. If so, I'll have a go at it.

> The best is to use the generic_fk from the forward kinematics package, then you can directly pass your urdf file.

Thanks for the suggestion, but what values should I use for `child_link` and `parent_link`?

maxspahn, 2023-10-10:
> Thanks for the suggestion, but what values should I use for child_link and parent_link?

`child_link` is the link of the end-effector you are concerned with and `parent_link` is usually the root link of the kinematic chain.

behradkhadem, 2023-10-12:
> `child_link` is the link of the end-effector you are concerned with and `parent_link` is usually the root link of the kinematic chain.

I understand it, but which one I should use for panda mobile (with gripper)?
https://github.com/maxspahn/gym_envs_urdf/blob/fdfe3d99a8f21c15581930e54d67ed738cf836da/urdfenvs/assets/mobile_panda/mobilePandaWithGripper.urdf#L8-L40
There are three baselinks: `base_link_x`, `base_link_y` and `base_link`. Which one is it?
I don't understand, why are we using `GenericUrdfReacher` and not `GenericDiffDriveRobot` (like we do with Albert)? Aren't they both non-holonomic? And it's confusing to choose the base link because out base is moving. Or am I misunderstood?

maxspahn, 2023-10-12:
> I don't understand, why are we using GenericUrdfReacher and not GenericDiffDriveRobot (like we do with Albert)? Aren't they both non-holonomic? And it's confusing to choose the base link because out base is moving. Or am I misunderstood?

The mobile panda is holonomic.

> I understand it, but which one I should use for panda mobile (with gripper)?

Try `world`. It should usually work for mobile robots.

behradkhadem, 2023-10-12:
> Try `world`. It should usually work for mobile robots.

Thanks! What should be the input of fk object? I can get meaningful results by using this:
`joint_position = observation['robot_0']['joint_state']['position'][:11]`
Is it correct? (fk input can't get the whole observation and says it should be 11-by-1 and not 12-by-one).

behradkhadem, 2023-10-12:
And also, I think the issue with grippers getting out of bounds (image above) is not from URDF file, but from pyBullet itself. Because URDF file has the limits (for the fingers) set correctly:

https://github.com/maxspahn/gym_envs_urdf/blob/fdfe3d99a8f21c15581930e54d67ed738cf836da/urdfenvs/assets/mobile_panda/mobilePandaWithGripper.urdf#L360C1-L394