maxspahn/gym_envs_urdf #271: Running multiple threaded envs in parallel

yj-Tang, 2024-08-14:
Hi Max, I’m currently trying to run multiple urdfenvs in parallel, as suggested [here](https://github.com/bulletphysics/bullet3/issues/2975). While everything runs smoothly overall, I'm encountering an issue with goal generation in the parallel environments, except for the first one.

In both the regular environments and the first parallel environment, the key for the goal is '1'. However, in the other parallel environments, the key unexpectedly changes to '0', and the goal position defaults to [0,0,0]. Could you clarify what the key represents in this context? Do you have any insights into what might be causing this issue?

Thank you for your time.
Best regards,
Yujie

maxspahn, 2024-08-16:
Hi @yj-Tang,

it is a very interesting use case for this repository, and I haven't played around with parallelization at all myself, so I am curious about this. And i am very eager to help here.

The indices of goals and obstacles have no meaning at all, so are simply the index provided by pybullet at creation of the instances. I hoped they would be unique, which does not seem to be true. For now, I have no quick fix.

Can you also provide me with a small example here?

Best, Max.

yj-Tang, 2024-08-19:
Hi Max,

I've forked this repo to [my GitHub repository](https://github.com/yj-Tang/gym_envs_urdf/tree/vectorized_envs). My changes are in the "vectorized_envs" branch. Here's a summary of the modifications:

1. Flattened the observation of burden and created a CustomizedFlattenObservation wrapper.
2. Implemented a vectorized environment using the flattened observation, allowing multiple environments to run independently.
3. Modified `gym_envs_urdf/urdfenvs/urdf_common/urdf_env.py` to support running multiple PyBullet environments in parallel.
4. Added a simple script `gym_envs_urdf/examples/test_parallel_env.py` to test the vectorized environment.

The vectorized environment appears to be running, but I've noticed an issue: the goal information (first 3 elements in the observation) seems incorrect.

I would greatly appreciate if you could give any comment on the goal information discrepancy.